matthewlai/JLCKicadTools #5: QFN-48 incorrect rotation in CPL file

atanisoft, 2019-12-29:
When placing a QFN-48 chip (ESP32-PICO-D4 as example) with pin 1 as right side bottom most pin the CPL puts it as bottom edge left most pin.

The KiCad all-pos csv shows it as:
```
"U6","ESP32-PICO-D4","QFN-48-1EP_7x7mm_P0.5mm_EP5.3x5.3mm_ThermalVias",163.576000,-91.440000,180.000000,top
```
generated CPL file shows it as:
```
U6,ESP32-PICO-D4,QFN-48-1EP_7x7mm_P0.5mm_EP5.3x5.3mm_ThermalVias,163.576000,-91.440000,90.0,top
```

matthewlai, 2020-01-04:
Can you verify and see if this is consistent with other QFN sizes? I used QFN-64 (I think) on a design, and the rotation was correct.

twam, 2020-01-05:
My SOP-18 was also incorrect. Suggest to include to pin count in the database file as well.

matthewlai, 2020-01-05:
It's a bit strange that different pin counts would have different canonical orientations. But definitely, we should include pin counts (that's why I used regex for pattern matching) if they actually have different orientations.

TobleMiner, 2020-05-09:
Ack, just had that happen to me, too. See #23

matthewlai, 2020-05-09:
I wonder if they just changed the canonical orientations for a bunch of footprints, and we should just nuke the rotations file and start new.

TobleMiner, 2020-05-09:
Idk, even EasyEDA does not get it right for all parts
[image]

We were trying to find a method for automating generation of the rotation database, but there does not seem to be a single source with consistent data.

Manawyrm, 2020-05-09:
Hm, I've looked over a couple of different ways to automatically get rotation information for all footprints now, but unfortunately wasn't successful (yet).

I've looked at the EasyEDA component API ( https://easyeda.com/api/products/C7171/components ) but it doesn't correspond to the SMT assembly rotation in any way. 

Automatically generating JLC orders and looking for the location of the red dot would work (in the json data), but that's really an evil hack and I'd prefer not to do that :smile: 

I've sent JLCPCB an e-mail asking for more information about the rotations and asked for a list of parts and their rotation or an export of all footprint preview images (including the red dot) to hopefully be able to generate a full list of parts (and their rotation).

matthewlai, 2020-05-09:
Yeah it would be great if we can get an authoritative reference!

closed 2022-08-30 by matthewlai