matthewlai/JLCKicadTools #9: Cannot handle unicode characters correctly

stuartpittaway, 2020-01-05:
Hello, great project!

If a component has a value with a unicode character like 10KΩ an error occurs

```
Traceback (most recent call last):
  File "generate_jlc_files.py", line 78, in <module>
    main()
  File "generate_jlc_files.py", line 71, in main
    if GenerateBOM(netlist_path, bom_output_path) and FixRotations(cpl_path, cpl_output_path, db):
  File "E:\source\JLCKicadTools\jlc_lib\generate_bom.py", line 65, in GenerateBOM
    out.writerow([c.getValue(), ",".join(refs), footprint, lcsc_part_number])
  File "c:\Python34\lib\encodings\cp1252.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\u03a9' in position 4: character maps to <undefined>

```

matthewlai, 2020-01-05:
Can you try changing this line: https://github.com/matthewlai/JLCKicadTools/blob/1d69e674744e53e115730ac511bd721200c0e8b0/jlc_lib/generate_bom.py#L30

to:
`f = open(output_filename, mode='w', encoding='utf-8')`

and see if that works?

stuartpittaway, 2020-01-05:
Yes that seems to work

matthewlai, 2020-01-05:
Cool thanks. I submitted the change.

closed 2020-01-05 by matthewlai