matthewlai/JLCKicadTools #13: Space before part number " C123456" is treated as a missing part

netkruzer, 2020-03-05:
FYI, I just had a few boards printed and assembled with JLC PCB. Parts that were. " C123456" were not placed, but ones that were "C123456" were fine.

Can you trim out the whitespace out of the front of the field.

The Python script is excellent!

matthewlai, 2020-03-05:
Can you test it for me?

in generate_bom.py, change
LCSC_PART_NUMBER_MATCHER=re.compile('^C[0-9]+$')
to
LCSC_PART_NUMBER_MATCHER=re.compile('^\s*C[0-9]+\s*$')

And thanks!

netkruzer, 2020-03-05:
I re-put the space in R2's part number and did your code change.

**Before code change:**
`Components R2, R6, R19, R14, R9, R24, R27, R32, R38 from same group do not all have LCSC part number C17414 set. Use --assume-same-lcsc-partnumber to ignore.`

**After code change:**
`Components R2, R6, R19, R14, R9, R24, R27, R32, R38 from same group have different LCSC part numbers:  C17414, C17414`

Checking the CSV, it also doesn't list R2 in the BOM and did not create the CPL file. Looks like it still needs another fix.

matthewlai, 2020-03-05:
Ah I see. Can you try reverting that, and change this line: https://github.com/matthewlai/JLCKicadTools/blob/817c56906da3d516d37b80e492bd518297990daf/jlc_kicad_tools/jlc_lib/generate_bom.py#L56

From:
`field_value = c.getField(field_name)`
To:
`field_value = c.getField(field_name).strip()`
?

Thanks

netkruzer, 2020-03-05:
That works!

Thanks. No error on running the script and both the BOM and CPL are created.

matthewlai, 2020-03-05:
Committed. Thanks for your help!

closed 2020-03-05 by matthewlai

netkruzer, 2020-03-05:
> Committed. Thanks for your help!

Thanks for making this script. I've sent 3 different assemblies to JLC and they've been great. So much better than soldering everything by hand.