matthewlai/JLCKicadTools #19: components being grouped across different designators

Shackmeister, 2020-05-04:
From how I see it, the tool should only try to group components if they are the same designator type (R1, R5, R10 etc) besides FP, value etc.

An NTC resistor with same value and FP as a normal resistor, gave me this warning:
"Components R5, TH1 from same group have different LCSC part numbers: C25744, C92371"
Is this a bug or by design?

matthewlai, 2020-05-04:
That's actually how KiCad's netlist parsing script groups components: https://github.com/matthewlai/JLCKicadTools/blob/6e19b39dea3fa3014f29e1715333116c6c34c6ad/jlc_kicad_tools/jlc_lib/kicad_netlist_reader.py#L651
https://github.com/matthewlai/JLCKicadTools/blob/6e19b39dea3fa3014f29e1715333116c6c34c6ad/jlc_kicad_tools/jlc_lib/kicad_netlist_reader.py#L338

I agree it makes sense to also use refdes. Happy for you to change it if you have a moment for a PR.

Shackmeister, 2020-05-04:
hmm currently it seems like by build uses the internal libraries for the Kicad modules.
Would you know how to force it to use the ones in the repo?

matthewlai, 2020-05-04:
Are you sure? The library is imported by path: https://github.com/matthewlai/JLCKicadTools/blob/6e19b39dea3fa3014f29e1715333116c6c34c6ad/jlc_kicad_tools/jlc_lib/generate_bom.py#L19

Though it's possible that if you installed with pip it may be using an installed copy. I would suggest uninstalling from pip, and just git clone the repo instead.

Shackmeister, 2020-05-05:
Fixed in https://github.com/matthewlai/JLCKicadTools/pull/20

closed 2020-05-05 by Shackmeister