matthewlai/JLCKicadTools #28: JLC have inconsistent rotations across *same* footprint

polwel, 2020-06-04:
I just needed to replace a diode (C95872 by C412437) because JLC went out of stock on the former. 
I now see that the new diode is incorrectly rotated, although I am confident that the rotation was correct with the other part number. I cannot verify now, because, well, JLC won't allow me to visualize any components that are out of stock...

Anyhow, this strongly suggests that not all the components of the same package are oriented the same way. It might just be an oversight on JLC's side.

This script should support a by-part-number rotation.

polwel, 2020-06-04:
Another example: C130204 and C7859, both TSSOP-24, are rotated differently.

klpeders, 2020-06-25:
IMHO, this is not a JLC issue, but rather a issue the standard used in their SMT machines (IEC61188 standard?). See https://forum.kicad.info/t/jlcpcb-bom-and-pick-and-place/19016

JLC them self have a longish blog post detailing how it works: https://support.jlcpcb.com/article/99-does-the-red-dot-means-pin-1-in-the-placement-previewer
[image]

> **About the “0” degree for the package**
> There is only one simple rule: Zero orientation for PCB library should match the component orientation in the packaging(tape/reel).

Accordingly, the idea that you can have a database of rotation by *footprint* doesn't work.
For example, ONSEMI details the rotation here: https://www.onsemi.com/pub/Collateral/BRD8011-D.PDF

Take for example SOT-89, ON have that in all four rotations:
[image]

rusefillc, 2023-05-16:
Just poking to see if there is any chance of progress? The issue seems real :)

rusefillc, 2023-05-16:
> will likely have basically 0 reusability
> Most components of the same footprint

Aren't those two statements contradicting each other? Most components of the same footprint are the same, we just need a way for the annoying exceptions!

rusefillc, 2024-05-09:
Hello from 2024.

matthewlai, 2024-05-09:
Sorry I forgot about this.

> Aren't those two statements contradicting each other? Most components of the same footprint are the same, we just need a way for the annoying exceptions!

Yes, I was referring to the case where we only do per-component and not per-footprint.

Per-footprint + exceptions for components may be possible, but I'm actually not sure how many exceptions there are, and if, for example, 10% of SOT-89 diodes are rotated one way, 90% are the other way, but the most popular ones are in the 10%, it's not clear what we should do, and it's also not clear how we can find out which one is more common to begin with, without checking a lot of parts.

This is getting really ugly, and I'm increasingly of the opinion that we should really just download the EasyEDA footprints (apparently there's a way to do batch downloads?), write a script to figure out how each component should be rotated, and then generate a per-component database for that, and retire the concept of using per-footprint rotations.

Unfortunately that's quite a lot of work, and I don't really have time to implement it at the moment. Contributions are of course welcome (if anyone wants to start by figuring out how to download the EasyEDA footprints, and write a script to determine the correct rotation. KiCad uses the IPC standard (pin 1 upper left), so we should be able to generate the per-component rotation file without having to map the footprints to KiCad footprints first.

axel-angel, 2024-05-29:
Hello there,
> if anyone wants to start by figuring out how to download the EasyEDA footprints
Anyone saw that this very useful script exist? https://github.com/uPesy/easyeda2kicad.py  using myself and it's very useful. Hope this helps.

matthewlai, 2024-05-29:
That is very cool! Yes, maybe that's the solution and this project doesn't really need to exist anymore.