matthewlai/JLCKicadTools #29: Tool Generates an empty BOM

Grottenolm85, 2020-07-06:
As described before. This Project creates an empty BOM but a working CPL
[Circuit.zip](https://github.com/matthewlai/JLCKicadTools/files/4878712/Circuit.zip)

matthewlai, 2020-07-11:
Hi @Grottenolm85 Sorry for the delay. I only just had a chance to look into this.

It looks like your schematic does not have LCSC part numbers (Cxxxxxx). Please see the blog post on how to add that.

closed 2020-07-11 by matthewlai

Grottenolm85, 2020-07-12:
Ok, i see, thanks. I usually just pick the parts from the catalogue in the JCB-website, rather than putting it into the schematic. I still need a properly formatted BOM though. Therefore it would be helpful to reorganize the kicad-Bom even without LCSC Part number.

matthewlai, 2020-07-12:
The problem is for most PCBs there will be components that cannot be assembled by JLC, so if we include every component group, there will be lines that need deleting in the common use case. That's why the script assumes that components that don't have a JLC part number are not going to be assembled by JLC.

matthewlai, 2020-07-12:
I have added an option to include all component groups even if there is no LCSC part number. I tried it on your circuit and it seems to work. You should be able to use it if you update JLCKicadTools through pip.

https://github.com/matthewlai/JLCKicadTools/commit/846c870eef394441375cb52749e4603012fb9960