matthewlai/JLCKicadTools #35: Addition of an extra database file

Shackmeister, 2020-10-22:
Having an option for an extra database file would be great for personal libraries.
This way I wouldn't need to flood the current database with own namescheme FP's :)

matthewlai, 2020-10-23:
I am travelling at the moment and don't have kicad set up on my laptop, but can you give this a try?
https://github.com/matthewlai/JLCKicadTools/tree/multiple_databases

You should be able to install with something like:
pip3 install git+https://github.com/matthewlai/JLCKicadTools.git@multiple_databases

sophiekovalevsky, 2020-12-31:
@matthewlai, I can confirm that https://github.com/matthewlai/JLCKicadTools/commit/8b895dfa05b56ec4dbfd149524c4c436df41d222 works right of the bat, either by setting absolute and relative db paths.

matthewlai, 2021-01-01:
@sophiekovalevsky Amazing thanks for confirming! I'll merge it.

closed 2021-01-01 by matthewlai