matthewlai/JLCKicadTools #44: Part being listed in BOM but not in CPL file

alek5k, 2021-01-06:
Hi, thanks for the great tool. I am having an issue generating the files - a particular part (C620322) shows up in the BOM file but not in the CPL file. There are no errors or complaints when running the script.

Relevant row in BOM:
"1825910-6","S1,S2,S3","SW_SMD-4P_Switch","C620322"

Nothing in the CPL file.

Your help in figuring out what the problem is would be much appreciated.

matthewlai, 2021-01-06:
Hi, can you share both the kicad project and the generated files?

alek5k, 2021-01-06:
I found that the error occurs when running File>Fabrication Outputs>Footprint Position File. The footprint in question does not get generated into the "all-pos.csv" file.  Not sure why... I've attached a minimal example with the footprint in question, and an additional footprint.

[JLCError.zip](https://github.com/matthewlai/JLCKicadTools/files/5777104/JLCError.zip)

alek5k, 2021-01-06:
Update: It seems as though the footprint itself was not marked as a Surface Mount component in Footprint Editor > Edit > Footprint Properties > Fabrication Attributes = Surface Mount. Even though it was showing up as Surface Mount by default when the footprint was placed.

closed 2021-01-06 by alek5k

matthewlai, 2021-01-06:
Glad you worked it out :). It's difficult unfortunately. Many boards will also have thru-hole or mechanical parts that aren't supposed to be placed, and they wouldn't be in the exported file either.