matthewlai/JLCKicadTools #45: Issue with extended character set

bg1729, 2021-01-25:
I had a component (C83451) where I had copy-pasted a description including the symbol for ohm (137M**Ω**) into the value field. It appears the JLC tools don't like this as the component appeared in the review tables when ordering but it was not shown on the graphical review. Changing the value in the schematic editor to remove the Ohm "Ω" character (and re-generating the BOM and CPL using your script) fixed the problem and the component was shown in the review window properly. 

Maybe consider looking for non-standard (non-UTF-8?) characters in the script and fixing or warning the user?

Thanks for the script - I have just placed my first order using it!

matthewlai, 2021-01-25:
Thanks. Can you post the generated files with and without the character?

bg1729, 2021-01-25:
I didn't keep the original files around so recreated them with a example project. I've attached BOMs and CPLs with and without the omega character.

The output from your script looks sensible to me and I expect the issue is at JLC's end - their web application that shows the components on the pads so you can verify placement seems to silently ignore lines with the omega / ohm character and probably other characters too. 

It took me a while to track down what was going on and I thought adding a warning to the user or throwing an error when extended characters were found would prevent others having the same issue. 

[no_ohm_test_bom_jlc.txt](https://github.com/matthewlai/JLCKicadTools/files/5869031/no_ohm_test_bom_jlc.txt)
[no_ohm_test_cpl_jlc.txt](https://github.com/matthewlai/JLCKicadTools/files/5869032/no_ohm_test_cpl_jlc.txt)
[with_ohm_test_bom_jlc.txt](https://github.com/matthewlai/JLCKicadTools/files/5869038/with_ohm_test_bom_jlc.txt)
[with_ohm_test_cpl_jlc.txt](https://github.com/matthewlai/JLCKicadTools/files/5869039/with_ohm_test_cpl_jlc.txt)

matthewlai, 2021-01-25:
Ah I see. Yeah that looks like a problem on JLC's end. It's not easy though - for example, your descriptions also have the ± character which is also non-ASCII but seems to work fine.

bg1729, 2021-01-25:
Yes - the plus / minus appears OK which is odd. I just tried a test project from scratch. All of the resistors here are the same 0402 part but with different "Value" fields in EEschema

[image]

[image]

R1 only goes missing from the preview 

[image]

Chinese characters in the tables

[image]

"Comment","Designator","Footprint","LCSC Part #"
"137MΩ","R1","R_0402_1005Metric","C25741"
"1K ±20%","R2","R_0402_1005Metric","C25741"
"1K","R3","R_0402_1005Metric","C25741"

Designator,Val,Package,Mid X,Mid Y,Rotation,Layer
R1,137MΩ,R_0402_1005Metric,105.955001,-87.500000,90.000000,top
R2,1K ±20%,R_0402_1005Metric,110.000000,-87.500000,90.000000,top
R3,1K,R_0402_1005Metric,115.000000,-87.500000,90.000000,top

matthewlai, 2021-01-25:
Thanks for the test! I think the Chinese character means it's not parsing as UTF-8, but some other encoding. Maybe the ohms sign is a problem here because it's at the end of the field and that interferes with their parsing?

bg1729, 2021-01-25:
No worries - happy to help. 

I agree some kind of parser bug seems a reasonable guess. Probably safest to stick to basic ASCII characters only in their assembly files I think!

matthewlai, 2021-01-25:
I've added it to the Readme. Thanks for the report! We may add a warning or something to the code once we have a better idea of what triggers this JLC bug.

closed 2021-01-25 by matthewlai

dongly, 2023-11-21:
Now, JLC is suport utf-8 file