matthewlai/JLCKicadTools #49: How to rotate

arifainchtein, 2021-03-02:
Hello,
First off thanks for the blog post and the script.  I got it working to the point where i can produce the needed files.
The problem that i have is that one of my components needs to be rotated by 90 degrees.  
I tried following your instructions but i cant find the ‘cpl_totations_db.csv’ file. You mentioned that is in the same
directory as the conversion script, but I cant find that either.

I am running on OSX, and while i can execute the command

 **jlc-kicad-tools /Users/arifainchtein/Data/DigitalStables/mcu/kicad_designs/GloriaField/GloriaField**

i dont know where the files are installed and therefore the cpl files generated dont have the corrections
.  Could you give me some suggestions?

thanks

arifainchtein, 2021-03-02:
Hi,
Well i figured it out.  I found the rotation file and copied to the project directory.  The when invoking the script, you pass the -d parameter with the path to the rotation file.

One thing i noticed is that if the footprint has space in the name, you need to replace them with _, or even easier, look at the cpl created already and see how it refers to the footprint.
You are right,  so far i have had to rotate 3 parts.  I will send you a pull request with the complete list of parts so you can add it to the main file. 
Evertyhing is working on this front.  Thanks again for your work.

closed 2021-03-02 by arifainchtein

matthewlai, 2021-03-02:
Thanks for the update. Glad you figured it out and sorry I didn't see this till now.

You are right that using "-d" is the easier way now. It didn't exist back when I wrote the blog post, so I'll go update that :).

You don't actually need to copy the rotation file. You can just start a new file with your corrections, and it will be parsed after the standard file (and overwrite entries in the standard file if an entry already exists).

I didn't know that KiCad replaces spaces with _. Thanks for that info.

If you don't want to fork and do a submit, feel free to also just copy and paste your corrections here and I can add it to the file directly :).

Good luck with the board!

arifainchtein, 2021-03-03:
> <reply@reply.github.com>
> <state_change@noreply.github.com>

Hi Matthew,
Thanks for the email.  Here are the modifications i made

"^SOIC127P798X216-8N",-90
"^Relay_DPDT_Omron_G6K-2F",-90
"^SW_DIP_SPSTx01_Slide_Copal_CHS-01B_W7.62mm_P1.27mm",-180
"^BatteryHolder_Keystone_1060_1x2032",-180

i will send you more as i make them

By the way, any suggestions as to why footprints that i created end up
in the bom but not in the cpl file?

thanks

[2]. 

Links:
------
[1]
https://github.com/matthewlai/JLCKicadTools/issues/49#issuecomment-789273996
[2]

matthewlai, 2021-03-03:
Added, thanks!

For footprints you create, you have to set the SMT flag so KiCad will export it in CPL.