matthewlai/JLCKicadTools #54: a mistake in the listing i gave you

arifainchtein, 2021-03-22:
Hi Matt,
I received the cbs last week and i have been working on the code.  Yesterday I started testing the relay and i could not get it to switch.  I checked that the NC was working but the NO did not work.  I reviewd every component of that part of the circuit and i think the culprit is the relay.  It is placed reverse. the coil side is 180 degrees opposite of it should be.

I had add it this line to the ritation file:

"^Relay_DPDT_Omron_G6K-2F",-90

because originally was placed perpendicular, I guess i should have one:
"^Relay_DPDT_Omron_G6K-2F",90

Just wanted to give you a head's up. I am working through the pcb and i expect to make the next version in a couple of weeks.  
Will send you an update once i have verified that this correction fixes the problem

thanks

matthewlai, 2021-03-23:
Thanks for the heads up. I have reverted them for now. Let me know if that works!

arifainchtein, 2021-03-23:
all the other ones are ok, the relay is the only wrong one.

closed 2022-08-30 by matthewlai