matthewlai/JLCKicadTools #60: Python Package Index & Semantic Versioning.

mholiv, 2021-04-05:
Have you ever considered using some for of semantic versioning and publishing this tool via pypi. It would lower the barrier on entry and would make it easier for users to update things. I would be willing to help here if you are up for it.

My thought process:
1. Separate rotation/other database from code.
2. Add `update` command where the latest database could be pulled from github
3. Use git actions to CI/CD to push tagged commits into Pypi.
4. Only push a tagged commit when code changes are made. Database changes can happen in the background.

What do you think?

matthewlai, 2021-04-05:
Hi @mholiv, thanks for the suggestion. I think that makes sense, but I know very little about how Python packaging/distribution works. @sophiekovalevsky added the current pip support.

@sophiekovalevsky what do you think about this proposal?

mholiv, 2021-04-05:
No problem. If this is an area where I can help I would be happy to contribute. I have experience in python packaging and would be happy to help if you are up for it.

sophiekovalevsky, 2021-04-05:
@mholiv firstable, thank you for time in trying to improve JLCKicadTools. 

@matthewlai about the proposal, I think that's a really great idea since it will make the app more pythonic. The changes that I made a couple of months ago were with that specific aim, that in a future we could publish it on pypi and definitively it will make easier for the community to access the new changes. 

@mholiv I would be happy to help you in anything you need to implement this. If you can create a PR with the implementation that you suggested, I will test it and give you feedback if there is something to change.

@matthewlai are you okey with this?

matthewlai, 2021-04-05:
@sophiekovalevsky yes, all sounds great to me, thanks!

closed 2022-08-30 by matthewlai