matthewlai/JLCKicadTools #63: Component not being replaced

arifainchtein, 2021-05-21:
Hi Matthew,
Here is the case.  
I made a project, did everything according to the instructions and generated the files.  I uploaded the gerners, the bom and the cpl to jlcpcb and discovered that two components where out of stock.
I went back chose replacements for those two components and made sure that all the places of those compnents had the new JLCPCB codes in the components in the sch.  
I then generated the bom and check the xml file to make sure that the new JLCPCB codes where in the xml.
Then when i ran the jlckicadtools script it kept using the old components code.
I cant seem to find where is the jlckicadtools is getting the previous code since there is nowhere in the kicad files

matthewlai, 2021-05-21:
Hi @arifainchtein,

I am not sure what could be causing this as the only place jlckicadtools gets the JLC part numbers is the XML file.

My guess is either you are checking the wrong output files, or the old XML file still exists somewhere and is being picked up.

arifainchtein, 2021-05-22:
Hey,
I am very sorry, you are correct.  There was an old xml file in the folder and i did not see it. 
Thanks

closed 2021-05-22 by arifainchtein