matthewlai/JLCKicadTools #64: Components in bom.csv but not in cpl

arifainchtein, 2021-06-04:
Hi Matthew,

I am having an issue that would love to hear your feed back on.  I generated the JLCPCB files and when i got the pcbs a few components were not placed on the pcb.  I contacted jlcpcb and they told me that those components were not in the files and that it why they had not been placed.  
Sure enough when i checked the files i discovered that they are in the generated bom file but not in the cpl file.  I am including the files.
For example, the following components are present in the bom but not in the cpl:

C10 C14 C18 C22 C27
J13 J29
J10
C27
C11 C15 C19 C23 C25

Do you have any suggestions as to what i could be doing wrong in my kicad files for the components not to appear in the cpl file?

Thanks

[valentino_2104.pro_bom_jlc.csv](https://github.com/matthewlai/JLCKicadTools/files/6601610/valentino_2104.pro_bom_jlc.csv)
[valentino_2104.pro_cpl_jlc.csv](https://github.com/matthewlai/JLCKicadTools/files/6601611/valentino_2104.pro_cpl_jlc.csv)

arifainchtein, 2021-06-04:
i just discovered one more thing.
It turns out that the kicad generated cpl file does not have the components either, here is the kicad generated cpl file.
If your script is using that file , it would explain why the components are not being included in the cpl generated file.
The question then is what am i doing wrong for kicad not to include those components in the cpl file.
One more piece of information is that all the components that do not appear in the cpl file are through hole components, while non of the smt components are missing.
Any thoughts?
[valentino_2104-all-pos.csv](https://github.com/matthewlai/JLCKicadTools/files/6601640/valentino_2104-all-pos.csv)

matthewlai, 2021-06-05:
Yes, if KiCad doesn't include the file in the output, the script won't have it.

For through hole components you need to click into the footprint and change "fabrication attributes" to "surface mount". It doesn't actually change the PCB. It just includes the footprint in the placement file.

Yes, it's pretty silly. Some background:
https://bugs.launchpad.net/kicad/+bug/1656688
https://forum.kicad.info/t/through-hole-parts-tht-in-position-files-missing/19927

arifainchtein, 2021-06-08:
Hi Matthew,
Just to let you know that indeed you were right.  When I changed the footprint to smd they now appear in the cpl file and your code picks them up.

I have modified a tool I created to work  with JLCKicadTools.  My tool [https://github.com/arifainchtein/kicadparser](url) will read the cpl file and generate an analysis of what references are in the net file but do not appear in the cpl, which is very usefull if you are using hand soldering components.

I will close this issue
Thanks

closed 2021-06-08 by arifainchtein