matthewlai/JLCKicadTools #70: Mounting on backside of PCB

Shackmeister, 2021-08-02:
Anyone having luck with exporting placement file for the backside? Mine seem to be completely off in the preview.
Not sure if this is script or JLC related

matthewlai, 2021-08-02:
I have never tried but I think on the KiCad CPL output you can choose whether x/y is mirrored. It could be something to do with that.

Shackmeister, 2021-08-02:
Tried again with the mirror X version, didn't seem to do anything

maikramer, 2021-09-07:
I have this problem also, mounting the back make the parts not to appear in the preview.

matthewlai, 2021-09-07:
Can you look at the csv output file and compare it to JLC's example files and see what's going on?

maikramer, 2021-09-07:
Even not showing in the preview, JLCPCB could manufacture my boards, but they need some manual corrections. generated csv is:
```
Designator,Val,Package,Mid X,Mid Y,Rotation,Layer
C1,100nF,C_0402_1005Metric,235.232000,-119.634000,180.000000,bottom
C2,10uF,C_0805_2012Metric,234.762000,-117.856000,180.000000,bottom
C3,10uF,C_0805_2012Metric,180.975000,-108.077000,90.000000,bottom
C4,100nF,C_0402_1005Metric,249.174000,-95.123000,180.000000,bottom
C5,47uF,C_1206_3216Metric,249.174000,-91.997000,-90.000000,bottom
C6,47uF,C_1206_3216Metric,183.564000,-85.598000,180.000000,bottom
C7,47uF,C_1206_3216Metric,183.564000,-83.185000,180.000000,bottom
C8,47uF,C_1206_3216Metric,183.515000,-92.329000,180.000000,bottom
C10,100nF,C_0402_1005Metric,197.485000,-77.724000,180.000000,bottom
C11,100nF,C_0402_1005Metric,197.485000,-127.508000,180.000000,bottom
D5,ZMM5V6,D_MiniMELF,183.289000,-94.742000,180.000000,bottom
D6,SS54,D_SMA,189.484000,-110.109000,180.000000,bottom
D7,SS14,D_SMA,178.500000,-125.500000,180.000000,bottom
F1,500mA,Fuse_0805_2012Metric,187.452000,-96.520000,90.000000,bottom
F2,500mA,Fuse_0805_2012Metric,189.865000,-96.520000,90.000000,bottom
F3,500mA,Fuse_0805_2012Metric,192.278000,-96.520000,-90.000000,bottom
F4,500mA,Fuse_0805_2012Metric,194.691000,-96.520000,-90.000000,bottom
F5,500mA,Fuse_0805_2012Metric,197.104000,-96.520000,-90.000000,bottom
F6,500mA,Fuse_0805_2012Metric,199.517000,-96.520000,90.000000,bottom
Q1,SS8050,SOT-23,224.856000,-122.506500,270.000000,bottom
Q2,SS8050,SOT-23,221.361000,-122.506500,270.000000,bottom
Q3,AO3400A,SOT-23,196.596000,-87.122000,0.000000,bottom
R1,330R,R_0805_2012Metric,208.531000,-102.362000,180.000000,bottom
R2,470R,R_0805_2012Metric,208.507500,-105.156000,180.000000,bottom
R3,330R,R_0805_2012Metric,208.531000,-107.950000,180.000000,bottom
R4,5.1K,R_0402_1005Metric,177.800000,-93.980000,90.000000,bottom
R5,5.1K,R_0402_1005Metric,177.800000,-89.789000,-90.000000,bottom
R6,10k,R_0603_1608Metric,251.333000,-94.297000,-90.000000,bottom
R7,10k,R_0603_1608Metric,228.410000,-123.444000,180.000000,bottom
R8,10k,R_0603_1608Metric,228.410000,-125.349000,180.000000,bottom
R9,2K,R_0402_1005Metric,178.562000,-99.314000,180.000000,bottom
R10,47R,R_0402_1005Metric,228.727000,-118.364000,0.000000,bottom
R11,47R,R_0402_1005Metric,228.727000,-119.634000,0.000000,bottom
R12,10k,R_0603_1608Metric,192.659000,-115.189000,0.000000,bottom
R13,2K,R_0402_1005Metric,180.975000,-112.014000,180.000000,bottom
R14,10k,R_0603_1608Metric,186.627000,-101.092000,180.000000,bottom
R15,5.1K,R_0402_1005Metric,178.816000,-119.634000,90.000000,bottom
R16,5.1K,R_0402_1005Metric,178.816000,-116.078000,-90.000000,bottom
R17,10k,R_0603_1608Metric,186.627000,-107.315000,180.000000,bottom
U1,MP6513GJ,MP6513GJ,197.612000,-80.010000,0.000000,bottom
U2,MP6513GJ,MP6513GJ,197.612000,-129.794000,0.000000,bottom
U3,ESP32-WROVER,ESP32-WROVER,236.616000,-91.291500,180.000000,bottom
U4,CH340C,SOIC-16_3.9x9.9mm_P1.27mm,236.347000,-123.952000,0.000000,bottom
U5,ME6211C33M5G-N,SOT-23-5,184.089000,-89.032500,270.000000,bottom
U6,TP4054,TSOT-23-5,177.546000,-110.871000,180.000000,bottom
U7,TXB0104RUT,Texas_R_PUQFN-N12,211.328000,-89.154000,180.000000,bottom
D1,3.3V,LED_1206_3216Metric,208.026000,-102.362000,0.000000,top
D2,5V,LED_1206_3216Metric,208.023000,-105.156000,0.000000,top
D3,IO2,LED_1206_3216Metric,208.026000,-107.950000,0.000000,top
D4,Carregando,LED_1206_3216Metric,177.546000,-99.314000,0.000000,top
J1,Bat,JST_XH_B2B-XH-A_1x02_P2.50mm_Vertical,177.997000,-103.628000,-90.000000,top
J2,JTAG,JTAG_2x05_P1.27mm_Vertical,206.629000,-114.808000,-90.000000,top
J3,USB-C-Power,CUI_UJC-HP-3-SMT-TR,180.147000,-92.000000,-90.000000,top
J4,MosFET,PinHeader_1x02_P2.54mm_Vertical,190.114000,-86.995000,90.000000,top
J5,I/O 5V,PinSocket_1x04_P2.54mm_Vertical,208.026000,-88.646000,0.000000,top
J6,H3,PinSocket_1x10_P2.54mm_Vertical,221.100000,-80.024814,90.000000,top
J7,H1,PinSocket_1x10_P2.54mm_Vertical,217.000000,-81.100000,0.000000,top
J8,H4,PinSocket_1x10_P2.54mm_Vertical,220.900000,-129.935000,90.000000,top
J9,H2,PinSocket_1x09_P2.54mm_Vertical,217.000000,-107.580000,0.000000,top
J11,Saída M1,JST_XH_B2B-XH-A_1x02_P2.50mm_Vertical,202.750000,-79.475000,0.000000,top
J12,Saída M2,JST_XH_B2B-XH-A_1x02_P2.50mm_Vertical,202.662000,-128.576000,0.000000,top
P1,USB-C-Data,USB_C_Receptacle_HRO_TYPE-C-31-M-12,177.642000,-117.856000,90.000000,top
Ref**,Val**,VDS,189.257547,-128.764788,0.000000,top
SW1,Reset,SW_SPST_CK_RS282G05A3,191.262000,-93.345000,0.000000,top
SW2,IO34,SW_SPST_CK_RS282G05A3,191.262000,-115.189000,0.000000,top
SW3,Chaves,SW_DIP_SPSTx02_Slide_9.78x7.26mm_W8.61mm_P2.54mm,191.729762,-104.310000,180.000000,top
X1,TinyProtoboard,TinyBreadboard,236.500000,-105.000000,0.000000,top
```

closed 2022-08-26 by Shackmeister