matthewlai/JLCKicadTools #72: Moving requirement for logzero to setup.py

kroesche, 2021-09-21:
Thank you for making this tool. It is very useful for generating JLC production files.

Is there a reason you have the two-step install, using pip to install requirements instead of just putting the requirement for `logzero` in the setup.py file? If not, then I would be happy to submit a PR to move this requirement into the setup.py.

matthewlai, 2021-09-22:
@sophiekovalevsky

sophiekovalevsky, 2021-09-22:
@matthewlai, thanks to ping, otherwise I couldn't catched. 

@kroesche, there is no particular reason and I think what you brought up is actually a great idea. I would be happy to review your PR.

kroesche, 2021-09-23:
I submitted PR https://github.com/matthewlai/JLCKicadTools/pull/73

closed 2021-09-24 by sophiekovalevsky