matthewlai/JLCKicadTools #82: Moving to Kicad-plugin?

Shackmeister, 2022-04-04:
Hi @matthewlai 

First, thanks for the great work :)
I noticed V6 imports all the extra fields from schematic (LCSC etc).
I guess this would make it possible to run the script from PCBnew, and avoid a bunch of the export steps.
Perhaps it could even be made as an actionplugin.
Would this make sense?

I do understand if the amount of work would not be worth the gain.

Kindest regards

matthewlai, 2022-04-05:
Hi @Shackmeister 

That does sound interesting and would be a good simplification. Unfortunately I'm very busy for the next 2 months so I am unlikely to have time to look into that before then.

Contributions are of course welcome :)

hattesen, 2022-12-11:
I might have a go at "plug-in-ifying" the current python scripts according to the [Python Plugin Guide](https://dev-docs.kicad.org/en/python/pcbnew/).
Do you guys have any suggestions for how the UI integration should appear? I will do my best to maintain backwards compatibility with the current CLI implementation. I'll develop in a clone on a separate branch and submit a pull request, initially as a MVP.

hattesen, 2022-12-12:
> I think if a plugin works well, there's probably no need to maintain a CLI version.

On the other hand, unless there is a good reason not to maintain the CLI interface, why not leave it in. It may well provide easier testing, debugging and automation than running as a plug-in.

easyw, 2023-09-19:
@matthewlai the CLI interface is extremely useful, IMO.
i.e. kicad-cli is a response to users need...