matthewlai/JLCKicadTools #90: Throw error if multiple files are present

Shackmeister, 2022-08-26:
Hi All
I have spent a good half hour debugging why one component was missing in the JLC bom.
Turns out I had a seperate export folder, which also had a valid .pos file.
Could we perhaps throw an error if there are multiple XML, pos files etc?
alternatively use the latest one.

These file inconsistencies can be quite dangerous IMO

Thanks for a great tool!

matthewlai, 2022-08-30:
Thanks.  Yes, I agree that's not desirable.

I've made the changes in the check_single_file_test branch, but don't have easy access to a computer to test it now. Would you mind testing it for me? You should be able to install it with something like -
pip install git+https://github.com/matthewlai/JLCKicadTools@check_single_file_test

mikebwilliams, 2022-09-09:
I got some errors:

-  Needs all netlist_path changed to netlist_paths, and cpl_path to cpl_paths. 
- Change f-strings like following
-                 f"Multiple netlist files found - {*netlist_paths,}. "
-                 f"Multiple CPL files found - {*cpl_paths,}. "

matthewlai, 2023-11-21:
Sorry I missed this earlier. Thanks for the testing @mikebwilliams . Merged now.

closed 2023-11-21 by matthewlai