matthewlai/JLCKicadTools #91: Error on Backside component placement

Shackmeister, 2022-08-31:
When having the following input from the POS file:
```
Ref,Val,Package,PosX,PosY,Rot,Side
"H1","Pogopin_PNP","Mill-Max_Pogopin_0906-8-15-20-76-14-11-0_PNP_DUMMY",16.335000,1.485000,180.000000,bottom
"H2","Pogopin_PNP","Mill-Max_Pogopin_0906-8-15-20-76-14-11-0_PNP_DUMMY",-13.365000,1.485000,180.000000,bottom
"H3","Pogopin_PNP","Mill-Max_Pogopin_0906-8-15-20-76-14-11-0_PNP_DUMMY",-16.335000,-1.485000,180.000000,bottom
"H4","Pogopin_PNP","Mill-Max_Pogopin_0906-8-15-20-76-14-11-0_PNP_DUMMY",-16.335000,1.485000,180.000000,bottom
"H5","Pogopin_PNP","Mill-Max_Pogopin_0906-8-15-20-76-14-11-0_PNP_DUMMY",4.455000,1.485000,180.000000,bottom
"H6","Pogopin_PNP","Mill-Max_Pogopin_0906-8-15-20-76-14-11-0_PNP_DUMMY",7.425000,1.485000,180.000000,bottom
"H7","Pogopin_PNP","Mill-Max_Pogopin_0906-8-15-20-76-14-11-0_PNP_DUMMY",13.365000,-1.485000,180.000000,bottom
```

I get the following output in the CPL file:
```
Designator,Val,Package,Mid X,Mid Y,Rotation,Layer
H1,Pogopin_PNP,Mill-Max_Pogopin_0906-8-15-20-76-14-11-0_PNP_DUMMY,16.335000,1.485000,0.000000,bottom
H2,Pogopin_PNP,Mill-Max_Pogopin_0906-8-15-20-76-14-11-0_PNP_DUMMY,13.365000,1.485000,0.000000,bottom
H3,Pogopin_PNP,Mill-Max_Pogopin_0906-8-15-20-76-14-11-0_PNP_DUMMY,16.335000,-1.485000,0.000000,bottom
H4,Pogopin_PNP,Mill-Max_Pogopin_0906-8-15-20-76-14-11-0_PNP_DUMMY,16.335000,1.485000,0.000000,bottom
H5,Pogopin_PNP,Mill-Max_Pogopin_0906-8-15-20-76-14-11-0_PNP_DUMMY,4.455000,1.485000,0.000000,bottom
H6,Pogopin_PNP,Mill-Max_Pogopin_0906-8-15-20-76-14-11-0_PNP_DUMMY,7.425000,1.485000,0.000000,bottom
H7,Pogopin_PNP,Mill-Max_Pogopin_0906-8-15-20-76-14-11-0_PNP_DUMMY,13.365000,-1.485000,0.000000,bottom
```

Please note H1 and H4 is suddently placed in the same position!

Shackmeister, 2022-08-31:
@purdeaandrei you seem to know something about the bottom placement algorithm :)

Shackmeister, 2022-09-11:
flip_x autodetect, is assuming all components are placed at a positive coordinate.
It's not unusual to have components placed on both sides of 0.
I would suggest we remove the autodetect, and add a suggested export setting in the documentation

boborjan2, 2022-09-14:
Could you detail what to set during export exactly? I also experience odd placement on the bottom. Thanks!

Shackmeister, 2022-09-14:
Actually the issue is also present on the top layer
Issue is present when part are placed on -X or +Y (in Kicad)